octolance/n2d #1: Tool is connecting to my mouse instead of the ESP8266

codingneko, 2019-05-27:
Hey, I'm just trying out this tool and I have my mouse connected on COM1, and it happens to be detected as a serial connection, when I try flashing the firmware on the ESP8266 it auto-selects COM1 but my ESP is on COM8...

Here is n2d's output:
[image]

Here is the Arduino IDE configuration I am using (ESP on COM8 as you can see):
[image]

mrvodka007, 2019-05-27:
**Hi codingneko,** 

Thank you for raising this issue. Unfortunately at the moment you will have to unplug your mouse while attempting to flash your device with N2D. I am working on an algorithm for detecting the right device and I hope to get it out within a week or two. I am also going to add a COM Port selection so you can manually choose your COM Port.

**mrvoda007**

mrvodka007, 2019-05-27:
Hello again, I have updated the program and it should be working fine now. If not, feel free to open another issue. Thanks :smile:

closed 2019-05-27 by mrvodka007

codingneko, 2019-05-27:
Awesome! that was fast :open_mouth: I'll have a look when I get home, thank you so much!

mrvodka007, 2019-06-13:
It sure was, I realized how simple of a fix it was. :)