octolance/n2d #2: Doesn't detect CH340 (ESP8266), even with manual input and disabled COM1 port, tries going to COM1

MrEdinLaw, 2019-06-19:


mrvodka007, 2019-06-19:
Have you tried installing the CH340 driver, if so and auto detection doesn't work, please raise an issue on esptool.py's github page because it's their software that programs the device and n2d is a gui. 
https://github.com/espressif/esptool

MrEdinLaw, 2019-06-20:
> 
> 
> Have you tried installing the CH340 driver, if so and auto detection doesn't work, please raise an issue on esptool.py's github page because it's their software that programs the device and n2d is a gui.
> https://github.com/espressif/esptool

I did install the driver, in the end i did it over another tool which had let me decide which COM port i want. The problem was particularly that this one did not listen if i put a different COM port than the ones given.

mrvodka007, 2019-06-20:
Okay this is unusual behaviour and I'll have a look at it. The issue can stay open for now untill the fix is released.

mrvodka007, 2019-06-20:
Hi @MrEdinLaw, I believe the issue has been resolved. The auto mode now works :) and manual mode is also working as far as I can tell, I only have one COM device (my nodemcu) so I can't really check if it's really working. Please find the updated version in the release folder or compile it yourself. Thanks for raising this issue, if there are still problems raise another since I am closing this one. -- **mrvodka007**

closed 2019-06-20 by mrvodka007