octolance/n2d #6: my oled deauther's screen just shows gibberish and won't let me press any buttons...

lesquieshia, 2019-12-12:


mrvodka007, 2019-12-24:
Sorry for the long delay, N2D isn't meant for OLED deauthers. It uses the generic deauther firmware that is supposed to work for most NodeMCU or Wemos D1 Minis. Not for OLED deauthers. Again, apologies for the delay, and if you wish to use N2D, download the correct firmware and set it as the flash file when prompted by N2D. Merry Christmas! 🎄

closed 2019-12-24 by mrvodka007

wisehackermonkey, 2020-02-13:
darn, had the same issue and the fix was:
Fallowing the tutorial from https://github.com/spacehuhn/esp8266_deauther/wiki/Installation

basically pip install esptool

esptool -p /dev/ttyUSB0 write_flash -fm dout 0x0000 .\deauther_2.1.0_DSTIKE_Wristband_Mini.bin
windows where  com7 is found from arduino ide
 esptool -p "COM7" write_flash -fm dout 0x0000 .\deauther_2.1.0_DSTIKE_Wristband_Mini.bin

Note: install the DSTIKE Deauther Wristband
[image]