octolance/n2d #14: board not recognized correctly

GastonLagaffe2013, 2020-05-25:
Salut,
I have a wifi deauther OLED 3.5 and tried to write the current bin to it with n2d.
Unfortunetely it recognizes it as a deauther 2.1 and loads the wrong bin leading to an unusable device.
Here is the output on the n2d:
`[11:57] Started Services
[11:57] This app uses the first espressif serial device, remove any other RS232 devices or try https://github.com/mrvodka007/n2dx
[11:57] Searching for devices . . .
[11:57] Device Found at: COM4
[11:57] Preparing to flash . . .
[11:57] N2D needs access to the internet. Please allow it if prompted.
[11:57] Downloading package: http://github.com/spacehuhn/esp8266_deauther/releases/download/v2.1.0/deauther_2.1.0_1mb.bin
[11:57] Download OK
[11:57] Download Complete
[11:57] Flashing Device . . .
[11:58] Flash Complete!
`
I use windows 10-1909-64bit and n2d v2

Ciao, Mathias

mrvodka007, 2020-05-25:
Hi Mathias. If you do a little bit more reading on my repositoy, I **do indeed** say that N2D is meant for **NodeMCU's and D1 Mini's** so this is not the tool to use for OLED deauthers. Instead you should try n2dX. It let you use a custom firmware file that you can download from the deauther repo directly.

closed 2020-05-25 by mrvodka007

GastonLagaffe2013, 2020-05-25:
Salut, thanks for the hint - it is greyed out on my screen so I overlooked it