octolance/n2d #15: The board just starts flashing.

A2umaa, 2020-05-30:
After the process is comlpete, the board just starts flashing red, untill I press the flash button again. No WIFI or anything.

mrvodka007, 2020-05-30:
Hmm. That's odd. Usually when something doesn't flash correctly it flashes blue. That seems to tell me that it could be a problem with your specific device as the firmware is trying to report a problem (either WiFi or SPIFFS), try asking SpaceHuhn or try again. It should and does work with most devices.

A2umaa, 2020-05-30:
I am using this device: AZDelivery NodeMCU Lua Amica Modul V2 ESP8266 ESP-12F WIFI Wifi Development Board. I have also tried other flashing methods, none work. It just continues to blink. Thanks for responding so quickly.

mrvodka007, 2020-05-30:
In that case it's likely the problem is with your specific device. Sorry about that.

closed 2020-05-30 by mrvodka007