octolance/n2d #16: Dstike Display problem

Strungup, 2020-06-07:
The displays on my mini OLED deauther and my dstike wrist band have quit working and are completely black.  The boards work, but no display.please help

mrvodka007, 2020-06-07:
Hi, N2D is not meant for DSTIKE deauthers. It is only meant for the NodeMCU and D1 Mini boards that don't have an OLED display. Try using n2dX or another flash tool and manually download your correct firmware from SpaceHuhns repository.

DangerousPony204, 2020-06-13:
go to web interface and enable display again under settings

Strungup, 2020-06-17:
I've enabled display and rebooted multiple times. No results. I could have accidentally screwed it up with my sub-noob arduino skills.

mrvodka007, 2020-06-23:
Once again, this isn't the version for DSTIKE - yes the firmware may have an option for it but if you want it to work correctly you need to flash an alternate image using another tool as N2D can't do this for you. This problem will be gone when Deauther V3 launches.

Strungup, 2020-06-26:
thank you

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2020-06-26 by mrvodka007