octolance/n2d #23: my device wont connect

stuckinaparadox, 2020-11-17:


stuckinaparadox, 2020-11-17:
dstike deauther v1 wont connect

realmrvodka42, 2020-11-27:
Hey @stuckinaparadox,

Sorry for the delayed response (and probably the answer) but I can't really do anything with "won't connect", because there are many causes. You can try a **shorter, high quality USB cable**; if that doesn't work, check that you have installed the correct **drivers**, or if you require remote assistance, let me know on Discord: **realmrvodka#4269.**

glebanaHACKER, 2020-12-05:
My device not connect. I use FTDI FT232RL chip.

realmrvodka, 2020-12-05:
Hi, I'm sorry you're having connection issues, have you tried installing FTDI drivers on your machine? As far as I'm aware FTDI devices are not 'officially' supported (listed in their docs) by the deauther project but I don't think that should be a problem.

If you need any further help, reply and I'll do my best to answer.

Thanks,

negassa, 2020-12-23:
Bonjour j'ai récemment acheté NodeMCU LUA WiFi basé sur la norme de carte de développement ESP8266 pour Arduino Compatible TE437. Et j'ai installer les drivers CH340g et il n'arrive pas à à être reconnu.
Pourriez-vous m'aider s'il vous plaît.

glebanaHACKER, 2020-12-23:
> 
> 
> Bonjour j'ai récemment acheté NodeMCU LUA WiFi basé sur la norme de carte de développement ESP8266 pour Arduino Compatible TE437. Et j'ai installer les drivers CH340g et il n'arrive pas à à être reconnu.
> Pourriez-vous m'aider s'il vous plaît.

English please

closed 2020-12-27 by realmrvodka