octolance/n2d #26: error

nobody446, 2021-03-13:
trying to flash my dstike watch and get error file not found failed to start flash, Esptool,exe is missing please replace the file and try again. i did downloaded again and same thing. im new to this ordered the watch and it came with nothing on it when it was supposed to have the latest deauther installed.

realmrvodka, 2021-03-20:
Did you check that `esptool.exe` is actually in the same folder as `n2dapp.exe`, if it is and it isn't working please let me know.

nobody446, 2021-03-20:
Sent from ProtonMail Mobile

This is what I get and everything works until I flash and then I get this error.

nobody446, 2021-03-20:
Sent from ProtonMail Mobile

Now when I flas it is saying esp tool is missing

realmrvodka, 2021-03-20:
> Did you check that esptool.exe is actually in the same folder as n2dapp.exe, if it is and it isn't working please let me know.

closed 2021-04-23 by realmrvodka