octolance/n2d #31: Is the N2DX firmware going to be re-released ?

FaisalAhmed123, 2021-09-24:
Is the N2DX firmware going to be re-released ?

pseudo8086, 2021-09-25:
Could you please clarify what you mean by firmware? N2DX was a test application I created a few years ago, it was more of a user-interface experiment than anything useful. It wasn't actual firmware to be flashed to an ESP8266.

FaisalAhmed123, 2021-09-25:
> Could you please clarify what you mean by firmware? N2DX was a test application I created a few years ago, it was more of a user-interface experiment than anything useful. It wasn't actual firmware to be flashed to an ESP8266.

The N2DX version made it a lot easier to flash the firmware, however it seems to have been deleted by the author, the only trace of it I can find is on this website, can't find it anywhere else anymore

https://blog.connectinut.co.uk/2021/02/20/kick-anyone-out-of-their-own-wifi-without-their-password-esp8266-deauther/

pseudo8086, 2021-09-25:
ND2X is no longer maintained and is quite out of date, plus like I said it wasn't designed to be for widespread use but rather a test program. I find it very interesting that you say N2DX was easier to use; if you've got any feedback regarding N2D22, that would be appreciated so it can be improved.

closed 2021-10-05 by pseudo8086