octolance/n2d #32: What happened to the old n2d

MD5-Hashm, 2021-11-01:
Title says most of it. What happened to the old (late 2019 or early 2020 I think) n2d with the nice ui

a4004, 2021-11-02:
Hello @MD5-Hashm
I'm glad you liked the user interface of the version from 2019-2020 but sadly it's no longer available on this repository because it's outdated and might not work as well as the latest version. If you're interested in the older version of the program you can try to reconstruct it from the commit history of this repo or see if any of [these](https://github.com/a4004/n2d/network/members) forks are what you might like to use or modify.

MD5-Hashm, 2021-11-02:
No problem, still love the project just wondering where it went! Keep up the great work and have a wonderful day.

closed 2021-11-02 by MD5-Hashm