octolance/n2d #33: Searching for your device …

Adrilafondu, 2023-03-24:
This programme doesn't seem to find my NodeMCU .

Is there a solution to this.
[image]

a4004, 2023-03-24:
Hi @Adrilafondu,

Sorry to hear you're having issues using the tool. Are you able to share a copy of the debug file for examination as I cannot see the image you tried to upload.

The debug file should be located in the same folder as the `N2D22.exe` file. It looks like this: `dump-xxxxxxxxxxxxxxxxxx.log`.

Thanks,

Alex

Adrilafondu, 2023-03-24:
Ok but i have 9 file .log it ok ?

a4004, 2023-03-24:
Yes that's not a problem.

Adrilafondu, 2023-03-24:
[dump-638152923517973206.log](https://github.com/a4004/n2d/files/11067319/dump-638152923517973206.log)

Adrilafondu, 2023-03-24:
i managed to send only 1 file

a4004, 2023-03-25:
> [dump-638152923517973206.log](https://github.com/a4004/n2d/files/11067319/dump-638152923517973206.log)

Thanks for that @Adrilafondu,

It seems you are missing the required drivers for your NodeMCU which is why the tool is unable to detect it. Please take a look at [this](https://github.com/a4004/n2d/wiki/Supported-Devices#driver-guide) wiki page (https://github.com/a4004/n2d/wiki/Supported-Devices#driver-guide) for more information on how to install the correct driver for your NodeMCU.

If you're not sure which driver to install after looking at the wiki, you can actually just install all 3 of the common drivers and Windows should automatically select the correct one when you connect the NodeMCU.

- https://www.silabs.com/documents/public/software/CP210x_Universal_Windows_Driver.zip
- https://sparks.gogo.co.nz/assets/_site_/downloads/CH34x_Install_Windows_v3_4.zip
- https://ftdichip.com/wp-content/uploads/2021/08/CDM212364_Setup.zip

Hope this helps,

Alex

Adrilafondu, 2023-03-25:
so the folder once downloaded must be extracted to the N2D22 ​​file?

a4004, 2023-03-25:
To install the CP210x driver do the following:

- Download the zip file from https://www.silabs.com/documents/public/software/CP210x_Universal_Windows_Driver.zip.
- Extract the files into a folder and enter the folder containing the files.
- Right click `silabser.inf` and choose `Install`.
- Follow any additional instructions.

To install the CH34x driver do the following:
- Download the zip file from https://sparks.gogo.co.nz/assets/_site_/downloads/CH34x_Install_Windows_v3_4.zip.
- Extract the files into a folder and enter the folder containing the files.
- Right click `CH34x_Install_Windows_v3_4.EXE` and choose `Run as Administrator`.
- Click on `Yes` at the User Account Control prompt.
- Choose `INSTALL` on the installer window.

To install the FTDI driver do the following:
- Download the zip file from https://ftdichip.com/wp-content/uploads/2021/08/CDM212364_Setup.zip
- Extract the files into a folder and enter the folder containing the files.
- Right click on `CDM212364_Setup.exe` and choose `Run as Administrator`.
- Click on `Yes` at the User Account Control prompt.
- Choose `Extract` in the installation wizard. A second window may appear after.
- Continue with the installation wizard by following the instructions.

After installing these drivers you may want to restart your PC to make sure Windows actually loads them when needed.

Adrilafondu, 2023-03-25:
[dump-638153464040252690.log](https://github.com/a4004/n2d/files/11068839/dump-638153464040252690.log)

Adrilafondu, 2023-03-25:
strange I followed your instructions because I have the CP210x driver but it still does not work
do you know the source of the problem?

a4004, 2023-03-25:
It looks like the device isn't being detected still. Can you verify whether or not Windows is able to detect it in device manager?
To do this:

1. Press [image] + `R` to open the `Run` dialogue.
2. Type `devmgmt.msc` 
[image]
3. Click `OK`.
4. You should see something like this:
[image]
5. Expand `Ports (COM & LPT)` to see the devices related to this.
[image]
6. You should see something like this. In my case the NodeMCU is COM3 but yours might be a different number.
7. If this is not showing or you see a yellow warning symbol, then there is a problem with the device or the driver.

More help available at the wiki: https://github.com/a4004/n2d/wiki/Troubleshooting.

Adrilafondu, 2023-03-25:
I tested all the cables in my house without success however when I connect to the pc the small led lights up but the peripheral managers cannot find it here is a photo of the usb controllers
[image]

Adrilafondu, 2023-03-25:
yes, I tried an ultimate cable and it finally worked

Adrilafondu, 2023-03-25:
[image]

Adrilafondu, 2023-03-25:
to be sure I did not screw up on the installation
[image]

Adrilafondu, 2023-03-25:
That's right ?

a4004, 2023-03-25:
Good to hear @Adrilafondu!

Yes if you are seeing the installation complete message that means everything should have worked. You may need to unplug and plug the device in again so the software starts then try connecting to the `pwned` WiFi network using your phone or other WiFi enabled device. 

In regards to your problem, sometimes the cable you use can cause it to not be detected as some of them might be power only (i.e. they do not enable the communication between the computer and the device) and for this serial over USB connection it is ideal if the cable is short in length so the communication works best.

Hope everything works out for you! The password to the `pwned` network should be `deauther`. 

If you have any other questions, feel free to open up another issue as I will be closing this one.

Alex.

closed 2023-03-25 by a4004